RflySim/RflyExpCode #1: 如果 脚本提示快捷方式生成失败

lix1990110, 2020-04-15:
GenerateShortcutCMD.bat 脚本路径错误导致的快捷方式生成失败
解决办法：用记事本打开或者其他编辑环境打开GenerateShortcutCMD.bat修改即可
mshta VBScript:Execute("Set a=CreateObject(""WScript.Shell""):Set b=a.CreateShortcut(a.SpecialFolders(""Desktop"") & strDesktop &""\PX4Msys2Toolchain.lnk""):b.TargetPath=""C:\PX4PSP\PX4Msys2Toolchain\msys2.exe"":b.WorkingDirectory=""C:\PX4PSP\PX4Msys2Toolchain"":b.Save:close")
mshta VBScript:Execute("Set a=CreateObject(""WScript.Shell""):Set b=a.CreateShortcut(a.SpecialFolders(""Desktop"") & strDesktop & ""\Python38Env.lnk""):b.IconLocation =""C:\PX4PSP\Python38\python.exe,0"":b.TargetPath=""C:\PX4PSP\Python38Env.bat"":b.WorkingDirectory=""C:\PX4PSP\Python38Scripts"":b.Save:close")
mshta VBScript:Execute("Set a=CreateObject(""WScript.Shell""):Set b=a.CreateShortcut(a.SpecialFolders(""Desktop"") & strDesktop & ""\eclipse.lnk""):b.TargetPath=""C:\PX4PSP\eclipse\eclipse.exe"":b.WorkingDirectory=""C:\PX4PSP\eclipse"":b.Save:close")
mshta VBScript:Execute("Set a=CreateObject(""WScript.Shell""):Set b=a.CreateShortcut(a.SpecialFolders(""Desktop"") & strDesktop & ""\FlightGear-F450.lnk""):b.TargetPath=""C:\PX4PSP\FlightGear 2016.1.2\FlightGear-F450.bat"":b.IconLocation =""C:\PX4PSP\FlightGear 2016.1.2\bin\fgfs.exe,0"":b.WorkingDirectory=""C:\PX4PSP\FlightGear 2016.1.2"":b.Save:close")
mshta VBScript:Execute("Set a=CreateObject(""WScript.Shell""):Set b=a.CreateShortcut(a.SpecialFolders(""Desktop"") & strDesktop & ""\3DDisplay.lnk""):b.TargetPath=""C:\PX4PSP\3DDisplay\3DDisplay.exe"":b.WorkingDirectory=""C:\PX4PSP\3DDisplay"":b.Save:close")
if not exist %userprofile%\Documents\Ogre\Ghadamon\ md %userprofile%\Documents\Ogre\Ghadamon\
copy C:\PX4PSP\3DDisplay\ogre.cfg "%userprofile%\Documents\Ogre\Ghadamon\ogre.cfg" /y
mshta VBScript:Execute("Set a=CreateObject(""WScript.Shell""):Set b=a.CreateShortcut(a.SpecialFolders(""Desktop"") & strDesktop & ""\CopterSim.lnk""):b.TargetPath=""C:\PX4PSP\CopterSim\CopterSim.exe"":b.WorkingDirectory=""C:\PX4PSP\CopterSim"":b.Save:close")
mshta VBScript:Execute("Set a=CreateObject(""WScript.Shell""):Set b=a.CreateShortcut(a.SpecialFolders(""Desktop"") & strDesktop & ""\QGroundControl-4.0.5.lnk""):b.TargetPath=""C:\PX4PSP\QGroundControl\QGroundControl.exe"":b.WorkingDirectory=""C:\PX4PSP\QGroundControl"":b.Save:close")

RflySim, 2020-04-15:
你好。快捷方式生成失败通常是Windows10的实时防护（或者其他杀毒软件）拦截了bat脚本执行；关闭杀毒软件后，手动点击GenerateShortcutCMD.bat即可生成快捷方式。
对于你列举的问题，能不能把具体的错误提示列一下。或者具体修改了哪个地方，解决了这个bug？

lix1990110, 2020-04-15:
脚本执行就是快捷方式无法创建，我看了下GenerateShortcutCMD.bat脚本创建桌面的路径有问题
Set b=a.CreateShortcut(a.SpecialFolders(""Desktop"")&amp;""\RflyTools\PX4Msys2Toolchain.lnk"")有问题改为Set b=a.CreateShortcut(a.SpecialFolders(""Desktop"") &amp; strDesktop &amp;""\PX4Msys2Toolchain.lnk"")就OK

RflySim, 2020-04-15:
为了避免用户桌面快捷方式混乱，我们的快捷方式现在默认生成到桌面的\RflyTools文件夹里面，请检查是否有这个文件夹，里面的快捷方式是否正常产生。我刚才更新了第二讲的PPT，将其中桌面快捷方式的位置强调了一下。

RflySim, 2020-04-19:
2020/04/18日之前下载的安装包确实存在这个bug，解决方法在桌面新建
RflyTools文件夹，再运行脚本文件。最新的安装包已经解决这个bug。

closed 2020-06-20 by RflySim