RflySim/RflyExpCode #2: 代码下载提示TypeError: 'str' object cannot be interpreted as an integer

lix1990110, 2020-04-15:
Traceback (most recent call last):
  File "C:\PX4PSP\Python38\px_uploader.py", line 731, in <module>
    main()
  File "C:\PX4PSP\Python38\px_uploader.py", line 606, in main
    fw = firmware(args.firmware)
  File "C:\PX4PSP\Python38\px_uploader.py", line 128, in __init__
    self.image.append('\xff')
TypeError: 'str' object cannot be interpreted as an integer

C:\Users\Administrator\Desktop\e3>
这个错误好像是Python38的问题该如何解决？

RflySim, 2020-04-15:
不好意思，是我这边的失误，升级了最新的Python38留下了兼容性bug。
你可以用记事本或编辑器打开"C:\PX4PSP\Python38\px_uploader.py"文件，将  self.image.append('\xff')   修改为 self.image.extend(b'\xff')
我这边会尽快修复这个bug，并发布新的安装包。

RflySim, 2020-04-16:
最新安装包已发布，这个bug已消除。

closed 2020-04-16 by RflySim