RflySim/RflyExpCode #4: 3Ddisplay打开后闪退

xulinliu, 2020-04-21:
在打开3Ddisplay这个软件的时候，还没有进去就会闪退。

RflySim, 2020-04-21:
首先请确认显示器屏幕分辨率是否大于等于1920*1080 (1080P)。如果分辨率过小可能会导致显示异常的现象。

其次，请确认你的桌面快捷方式是否成功生成。如果没有成功生成可能是前几天安装后脚本执行失败bug所致（4月18日之后的安装包已经解决这个bug）。
这种情况的解决方法：在桌面新建一个RflyTools文件夹，再双击运行C:\PX4PSP\GenerateShortcutCMD.bat文件，点击桌面上的RflyTools文件夹内的3DDisplay快捷方式，看是否能正常启动。

如果还是无法解决，请继续留言。

xulinliu, 2020-04-21:
> 首先请确认显示器屏幕分辨率是否大于等于1920 * 1080（1080P）。如果分辨率过小可能会导致显示异常的现象。
> 
> 如果，成功确认可能是前几天安装后脚本执行失败bug所致（4月18日之后的安装包已经解决这个bug）。
> 这种情况的解决方法：在桌面新建一个RflyTools文件夹，再双击运行C：\ PX4PSP \ GenerateShortcutCMD.bat文件，单击桌面上的RflyTools文件夹内的3DDisplay快捷方式，看是否能正常启动。
> 
> 如果还是无法解决，请继续留言。

你好，我已安装该方法进行了操作，分辨率设置为了1080P，桌面也产生了快捷方式，但是打开3Ddisplay软件，在启动软件过程中，仍会闪退。

RflySim, 2020-04-21:
你好。你尝试运行一下安装包的RFLYExpPspSoftwarePack\4.HILApps\MSVCP_2019.07.20_X64.exe修复文件，看能不能解决。如果还是不能解决，换一款1080P 16:9的屏幕试试。
如果还是无法解决，可能是电脑操作系统的问题。请提供你的操作系统信息，我们会尝试修复此bug。

最后，我们近期会发布一款备用三维引擎软件，请关注我们的https://flyeval.com/course.html页面的FAQ标签。目前只能靠更换电脑来解决。

xulinliu, 2020-04-21:
> 你好。你尝试运行一下安装包的RFLYExpPspSoftwarePack\4.HILApps\MSVCP_2019.07.20_X64.exe修复文件，看能不能解决。如果还是不能解决，换一款1080P 16:9的屏幕试试。
> 如果还是无法解决，可能是电脑操作系统的问题。请提供你的操作系统信息，我们会尝试修复此bug。 [...]

你好：
      我已按操作进行修复文件，但仍会闪退；我的屏幕为1080P 16:9。
      以下是我的电脑信息：
      处理器：Intel(R)  Xeon(R) CPU E5-2643 v4 @3.40GHz
      Windows版本：Window10企业版  
       版本号： 1909
      操作系统版本：18363.418

xiangyingkun, 2020-09-07:
我的3DDisplay也出现闪退

xiangyingkun, 2020-09-07:
@RflySim

ChenZiBiao2000, 2021-03-03:
我的3Ddisplay打开后会自动降低分辨率，而且应用界面还产生一横一横的切片，这个怎么解决