RflySim/RflyExpCode #8: 网站上e3.1代码中传感器的标定中求加速度计的参数是6个，而课件Lesson07Sensor and Measurement Model求解出的是9个参数，方便发一下求解9个参数实验代码和和参考的文章吗？

DaMeng59, 2020-05-25:


WANGshouming4937, 2020-05-26:
6个参数或是9个参数在实现上区别不大。
1.修改模型函数 calFunc.m
 kx = p(1);
    ky = p(2);
    kz = p(3);
    bx = p(4);
    by = p(5);
    bz = p(6);
    phi = p(7);
    theta = p(8);
    psi = p(9);

    Ta = [1 psi -theta;
             -psi 1 phi;
             theta -phi 1];
    Ka = [kx 0 0;0 ky 0;0 0 kz];
    ba = [bx by bz]';
    m = length(x);
    Y = zeros(1, m);
    for i = 1:m  
        Y(i) = norm(Ta*Ka*(x(1:3, i) + ba));
    end
2.修改参数的初始化值 calLM.m
p0 = [1 1 1 0 0 0 0 0 0]';
p_init = [1.0 1.0 1.0 0.1 0.1 0.1 0.001 0.001 0.001]';

DaMeng59, 2020-05-26:
> 6个参数或是9个参数在实现上区别不大。
> 1.修改模型函数 calFunc.m
> kx = p(1);
> ky = p(2);
> kz = p(3);
> bx = p(4);
> by = p(5);
> bz = p(6);
> phi = p(7);
> theta = p(8);
> psi = p(9);
>
> ```
> Ta = [1 psi -theta;
>          -psi 1 phi; [...]
可以提供一下LM算法参考的哪篇文献或书籍吗？

DaMeng59, 2020-06-09:
利用Ch7Date的加速度数据所得图像与Lesson07Sensor and Measurement Model（http://rfly.buaa.edu.cn/course/ch/Lesson07V2.pdf）中的pp13图像完全不同,请问是数据的原因，还是其他？