RflySim/RflyExpCode #11: 安装软件包出现错误提示，无法安装

ivory1989, 2020-05-29:
首次安装为matlab2018b，安装以后，无法打开coptersim软件，然后更换为matlab2019b出现安装错误提示最初保存为codertarget.pixhawk.Cmake_Utiles.Px4PSP_CMake_Class的变量Px4PSP_CMakeInfo无法实例化对象，将以int32位西农港城读取
>In PX4InstallScript
此类型的变量不支持使用点进行索引
出错PX4InstallScript
请问该如何解决

RflySim, 2020-05-29:
关闭MATLAB，删除【我的文档】\MATLAB\Add-Ons\Toolboxes目录下的“PX4 PSP”文件夹，再重新安装可解决“Px4PSP_CMake_Class”变量问题。无法打开Coptersim软件一般都是系统的问题，目前提供信息不足，无法给出解决方案。通常需要重装系统或换电脑再尝试。

ivory1989, 2020-05-29:
已按您说的删除了PX4 PSP文件夹，弹出了安装框，但是窗口里又出现了新的报错，请问应该如何解决？

无法解析名称 matlab.internal.WebwindowManagerInterface。

出错 matlab.internal.cef.webwindowmanager (line 18)
          newObj.InterfaceObject = matlab.internal.WebwindowManagerInterface;

出错 matlab.internal.cef.webwindowmanager.instance (line 27)
            obj = matlab.internal.cef.webwindowmanager();

出错 matlab.internal.webwindowmanager/get.windowList (line 116)
            implList = matlab.internal.cef.webwindowmanager.instance.windowList;

出错 matlab.internal.webwindowmanager/findAllWebwindows (line 91)
            list = obj.windowList;

出错 matlab.ui.internal.dialog.DialogUtils.disableAllWindowsSafely (line 24)
            webWindows = matlab.internal.webwindowmanager.instance.findAllWebwindows();

出错 inputdlg (line 363)
  c = matlab.ui.internal.dialog.DialogUtils.disableAllWindowsSafely();

出错 PX4InstallScript

ivory1989, 2020-05-29:
> 关闭MATLAB，删除【我的文档】\MATLAB\Add-Ons\Toolboxes目录下的“PX4 PSP”文件夹，再重新安装可解决“Px4PSP_CMake_Class”变量问题。无法打开Coptersim软件一般都是系统的问题，目前提供信息不足，无法给出解决方案。通常需要重装系统或换电脑再尝试。
已按您说的删除了PX4 PSP文件夹，弹出了安装框，但是窗口里又出现了新的报错，请问应该如何解决？

无法解析名称 matlab.internal.WebwindowManagerInterface。

出错 matlab.internal.cef.webwindowmanager (line 18)
newObj.InterfaceObject = matlab.internal.WebwindowManagerInterface;

出错 matlab.internal.cef.webwindowmanager.instance (line 27)
obj = matlab.internal.cef.webwindowmanager();

出错 matlab.internal.webwindowmanager/get.windowList (line 116)
implList = matlab.internal.cef.webwindowmanager.instance.windowList;

出错 matlab.internal.webwindowmanager/findAllWebwindows (line 91)
list = obj.windowList;

出错 matlab.ui.internal.dialog.DialogUtils.disableAllWindowsSafely (line 24)
webWindows = matlab.internal.webwindowmanager.instance.findAllWebwindows();

出错 inputdlg (line 363)
c = matlab.ui.internal.dialog.DialogUtils.disableAllWindowsSafely();

出错 PX4InstallScript