RflySim/RflyExpCode #18: 3D DISPLAY app is no displayed correectly

SebastienFerrero, 2020-06-09:
[image]

RflySim, 2020-06-10:
Please close all antivirus programs, and run C:\PX4PSP\GenerateShortcutCMD.bat again by double-clicking it.

SebastienFerrero, 2020-06-11:
Thank you for the answer. I did that but the problem is still persisting.
[image]
[image]

RflySim, 2020-06-11:
You can create a folder with name "RflyTools" on the Desktop, then run the C:\PX4PSP\GenerateShortcutCMD.bat script. This is a bug of our old installation package, and we have corrected it in our latest download link.

SebastienFerrero, 2020-06-11:
It's working properly now. xie xie. I would say that you are doing great work in your lab.

I send a message to the e-mail asking about the price of the advanced features of the 'RflySim Advanced Version' but I still did not get an answer.could you please check.
[image]

RflySim, 2020-06-11:
Sorry, we did not reply in time . We are working hard to test the advanced versions to ensure stability and reliability, which is delayed by the COVID-19 situation. The test work will be finished within two weeks, then we will send you an email about your concern. 
The price will be cheap, because we want more people to use it and we also need some money for sustaining development demand. We have planned many wonderful new functions and experimental courses about swarm/vision/AI control for different types of unmanned vehicle systems. You can pay attention to the update of our website.

SebastienFerrero, 2020-06-11:
Ok, got it. Good luck. You are the best. Looking forward for the new features. ZAI JIAN...

RflySim, 2020-09-08:
Please try the backup 3D program:  https://bhpan.buaa.edu.cn:443/link/38B0A4213A98A67CFD96B1D9D41D021C