RflySim/RflyExpCode #19: External mode with the telemetry radios for Pixhawk autopilot?

SebastienFerrero, 2020-06-11:
Hello;
I have been reading this note from your awesome work:

**Important Note:** Recent testing has confirmed that external mode does not operate correctly with the telemetry radios. Data comes in very sporadically and in long intervals. We are looking into this and will provide an update to address the issue, in the mean-time, please use USB for serial connection for external mode.

Please, I would like to know if this problem has been solved or not yet. xie xie. If yes, please where can I find the updates?

RflySim, 2020-06-12:
We don't recommend to use "External Mode", because it is unstable and we have no plan to update it. In the latest update, we will support to send output signals of interest to a uORB block, then you can observe the signals in QGC during both simulations and real flight tests.

SebastienFerrero, 2020-06-12:
@RflySim  Thank you so much for your clarification. I really appreciate your hardworking.