RflySim/RflyExpCode #21: Pixhawk 4 from Holypro

EL-Wadi, 2020-07-08:
Dear Sir
I really appreciate your great effort and your valuable contribution to the UAV field
I was trying to work with Pixhawk 4 (fmu v5). I was able to build and upload the firmware for the example  in 8.3 folder.  I have changed the  build command to (px4fmu v5_defualt)
 However, the  computer could not recgnize the pixhawk board after that until I launch the QGroundcontrol and replace teh custome made firmware by the original one. 
Does your package work only with Pixhawk 1  (FMU v3). 
I also wat to know the price of the advanced version of your package.
regards;
[image]

RflySim, 2020-07-08:
Our platform support all commands in the Firmware\cmake\configs folder which includes fmu-v5, but the PX4-1.7.x firmware for fmu-v5 does not support S.Bus RC due to driver problems. Currently, you can try to replug your Pixhawk 4 and use a PPM RC receiver.
For your problem, we will release a new installation package including PX4-1.8 firmware, which will support fmu-v5 perfectly. So please wait for a few days, and I will reply in this issue when the installation package is updated.

RflySim, 2020-07-11:
Hello, we have release a new installation iso file, please go to here https://rflysim.com/download to acquire the latest download link. Now we have add the PX4 1.8 firmware into the package, which should work fine for fmu-v5.

EL-Wadi, 2020-07-13:
Thanks for the new package. I tried the e8_3_Hill example and it is uploaded to the bord without any problem. However, when I tried to upload my controller with ADC inputs, the firmware was uploaded but the computer could not recgnize the board after that. To be able to recignize it, I have to use the GGroundcontrol software to upload the orgininal firmware
When I replaced the ADC block by constants, the firmware was also uploaded and the computer can recgnize the board!
So, it seems tha the ADC block creates a problem with the pixhawk 4 from holybro
[image]

RflySim, 2020-07-15:
Have you tried MATLAB 2017b or Other type of Pixhawk (e.g., Pixhawk 1)?  Currently, we haven't modified and tested the ADC block because it is not main content in our course experiments and textbook. Thank you for the report, we will try to correct this bug in our latest release.

EL-Wadi, 2020-07-15:
Unfortuately, I do not have Pixhawk 1 right now. I ordered it. I will check when I get it.
I thought that pixhhawk 4 will have better performnace so i started with it. 
regards

EL-Wadi, 2020-07-25:
Recently, I received Pixhawk 1 from mrobotics, The problem still exists. if I upload the firmware from e8.3 Hill example, I got no problem. If I just add the ADC block and upload the firmware, the board will appear for few seconds on the com port then disappear.
After that if the board will disappear even with uploading the firmware from example 8.3 Hill. 
I have to flash the board with the stable firmware using Qgroundcontrol and then I can upload the example e8.3 Hill and the board will work normally.!!!!
[image]