RflySim/RflyExpCode #22: Advanced Version

EL-Wadi, 2020-07-08:
Dear Sir
Does the advanced version support  HIL for systems that include Aanalog to Digital signals beside the other sensors in the pixhawk autpilot.
[image]

RflySim, 2020-07-08:
We haven't test this function yet, you can post you requirements in more detail. Where should the ADC signals come from? the simulated signal in CopterSim ? or the real analog (or digital) signals on Pixhawk ADC port.

In our advanced platform, you can use Simulink/C++/Python to send a uORB message to Simulink controller, maybe you can use this signal as the ADC output to simulate other sensors.

EL-Wadi, 2020-07-08:
Thanks for your reply. 
The ADC signals should come from the simulink model of the quadrotor