RflySim/RflyExpCode #23: 遥控器试飞一切正常，编译结束后upload反复插拔仍然识别不到BootLoader

W20162021, 2020-07-17:
在进行硬件在环仿真时（Pixhawk2.4.8），按照流程配置，遥控器试飞，均正常，
但使用simulink编译成功，进行PX4Upload时，反复插拔多次仍然无反应。是否有解决方案

Loaded firmware for 9,0, size: 1481421 bytes, waiting for the bootloader...
If the board does not respond within 1-2 seconds, unplug and re-plug the USB connector.
Attempting reboot on COM5 with baudrate=57600...
If the board does not respond, unplug and re-plug the USB connector.

RflySim, 2020-07-18:
你好，可能是电脑上PX4驱动没有安装好，请尝试运行RFLYExpPspSoftwarePack\4.HILApps\px4driver.msi修复一下。如果QGC可以烧入.px4格式固件，但是Simulink自动下载功能无法使用，你可以手动用QGC将Simulink生成.px4固件烧入Pixhawk，文件目录在C:\PX4PSP\Firmware\build\px4fmu-v3_default\px4fmu-v3_default.px4

BqLion, 2021-08-13:
> 你好，可能是电脑上PX4驱动没有安装好，请尝试运行RFLYExpPspSoftwarePack\4.HILApps\px4driver.msi修复一下。如果QGC可以烧入.px4格式固件，但是Simulink自动下载功能无法使用，你可以手动用QGC将Simulink生成.px4固件烧入Pixhawk，文件目录在C:\PX4PSP\Firmware\build\px4fmu-v3_default\px4fmu-v3_default.px4

此回答解决了我的问题，感谢。

BqLion, 2021-08-13:
> 你好，可能是电脑上PX4驱动没有安装好，请尝试运行RFLYExpPspSoftwarePack\4.HILApps\px4driver.msi修复一下。如果QGC可以烧入.px4格式固件，但是Simulink自动下载功能无法使用，你可以手动用QGC将Simulink生成.px4固件烧入Pixhawk，文件目录在C:\PX4PSP\Firmware\build\px4fmu-v3_default\px4fmu-v3_default.px4

此回答解决了我的问题，感谢。
修复程序最新的目录为：
安装目录\drivers\px4driver.msi