RflySim/RflyExpCode #25: 生成代码并编译的时候出错

quande1314, 2020-07-23:
### Failed to generate all binary outputs.
Caused by:
Validation error(s): 
### Validating other build tools ...
Unable to locate build tool "Pixhawk Toolchain C Compiler": echo
Unable to locate build tool "Pixhawk ToolchainC Pre-Linker": echo
Unable to locate build tool "Pixhawk Toolchain Archiver": echo 

可生成代码，编译出错。我安装完成之后，附加功能不显示PSP支持包，但可以使用。

RflySim, 2020-07-23:
你好。你是否使用的是最新版的安装包？请尝试下载安装最新版软件。如果还有问题，请把编译信息全部贴出来，目前你的信息没法判断哪里出错了。

Walterove, 2020-08-02:
我也遇到这个问题了。悲伤
[image]

WANGshouming4937, 2020-08-02:
> 我也遇到这个问题了。悲伤
> [image]

看你的报错提示，应该是logger文件有问题，少了一个分号。

kcx064, 2020-08-02:
在Matlab function中如果出现了两个向量直接相加或者相减，就会出现这个问题。你检查下，你的代码。或者通过注释的方式逐步排查。-------- 原始邮件 --------发件人： Walterove <notifications@github.com>日期： 2020年8月2日周日 中午11:02收件人： RflySim/RflyExpCode <RflyExpCode@noreply.github.com>抄送： Subscribed <subscribed@noreply.github.com>主    题： Re: [RflySim/RflyExpCode] 生成代码并编译的时候出错 (#25)
我也遇到这个问题了。悲伤

Walterove, 2020-08-02:
> > 我也遇到这个问题了。悲伤
> > [image]
> 
> 看你的报错提示，应该是logger文件有问题，少了一个分号。

谢谢您！解决了，之前添加log的topic，少加了一个分号。

Walterove, 2020-08-02:
好的，谢谢您。确实是代码里面的问题，已经解决了

zz-1907, 2020-09-22:
我也遇到了类似的问题，该怎么解决呢？
v2/NuttX/nuttx-export/arch/chip -ID:/PX4PSP/Firmware/build/nuttx_px4fmu-v2_default/px4fmu-v2/NuttX/nuttx-export/arch/common -ID:/PX4PSP/Firmware/build/nuttx_px4fmu-v2_default/external/Install/include -IF:/360MoveData/Users/pc dell/Documents/MATLAB/Add-Ons/Toolboxes/PX4PSP/code/include -o rtiostream_utils.o G:/toolbox/coder/rtiostream/src/utils/rtiostream_utils.c
gmake: *** No rule to make target `F:/360MoveData/Users/pc', needed by `PX4_TaskControl.o'.  Stop.
D:\PX4PSP\example_test\e0\e0\2.PSPOfficialExps\px4demo_attitude_system_ert_rtw>echo The make command returned an error of 2
The make command returned an error of 2
D:\PX4PSP\example_test\e0\e0\2.PSPOfficialExps\px4demo_attitude_system_ert_rtw>An_error_occurred_during_the_call_to_make
'An_error_occurred_during_the_call_to_make' 不是内部或外部命令，也不是可运行的程序
或批处理文件。
### Creating HTML report file px4demo_attitude_system_codegen_rpt.html
警告: Validation warning(s): 
Skipped the existence check for 'Invoke Make'. Cannot determine the utility that needs to be checked for existence from the build tool command '$(call CALLPX4MK)'.
### Build procedure for model: 'px4demo_attitude_system' aborted due to an error.
Error(s) encountered while building "px4demo_attitude_system":
### Failed to generate all binary outputs.
Caused by:
Validation error(s): 
### Validating other build tools ...
Unable to locate build tool "Pixhawk Toolchain C Compiler": echo
Unable to locate build tool "Pixhawk ToolchainC Pre-Linker": echo
Unable to locate build tool "Pixhawk Toolchain Archiver": echo