RflySim/RflyExpCode #28: Error in Building "AttitudeControl_FLY"

EL-Wadi, 2020-08-04:
I got this error when I tried to build the model "AttitudeControl_FLY" in e5.4 folder
I followed your instructions and replaced the three files (costom_attctrl_e5, CMakeLists, and logger)  by the given files in the folder "PSPfile" .
After that replacement, I got the same error when I tried to compile any other Hill model in your package..
Everything is OK when I returned to the original firmware files (CMakeLists, and logger)
I am using windows 10 with Matlab 2017 and your latest package  (RflySim V2.29 - 2020/07/08), firmware 1.7

[image]
[image]

WANGshouming4937, 2020-08-04:
You can try to modify the files instead of replacing with the given files.

EL-Wadi, 2020-08-04:
You have mentioned in your PPT slides to modify "logger.cpp" by adding “add_topic(“costom_attctrl_e5”, 4)” to the bottom line of “add_common_topic()” . There is no thing called add_common_topic(). There is add_default_topic(). Is this what you mean?

[image]

WANGshouming4937, 2020-08-09:
What you did is right, and you can modify the number as your wish. "add_common_topics()" appears in the older version, and we will correct the problem.

closed 2020-08-09 by EL-Wadi