RflySim/RflyExpCode #32: What is the best solution that you offer for simulation and implementation of FORMATION TRACKING CONTROL OF MULTIPLE QUADROTOR UAVs?

SebastienFerrero, 2020-08-19:
Dear flyeval group,

First, I would like to say congratulation for the new website https://rflysim.com/en/0_Start/WhatIsRflysim.html. That is really useful and important.
Second, please I would like to ask about what is the best simulation and implementation method that you offer for academic research regarding FORMATION TRACKING CONTROL OF MULTIPLE QUADROTOR UAVs? We hope to implement and test a distributed formation tracking control protocol on a swarm of UAVs.

Thank you, keep doing nice work, we are following your updates...
Sincerely,

RflySim, 2020-08-19:
Our RflySim advanced platform allows you to control UAV swarm through Simulink (or C++/Python) in both SIL/HIL simulations and indoor/outdoor flight tests. Here is a brief introduction on our website [https://rflysim.com/en/4_Pro/UAVSwarmControl.html](url) .  You can send e-mails to rflysim@163.com to get  further information and price sheet.

SebastienFerrero, 2020-08-19:
Thank you for your replay

mechaliomar, 2020-09-11:
Hello, please from where can I get this Simulink file?
[image]