RflySim/RflyExpCode #37: 编译Exp4_AttitudeSystemCodeGen_old时出错

LILILIHANG, 2020-10-19:
### Starting build procedure for model: Exp4_AttitudeSystemCodeGen_old
### Build procedure for model: 'Exp4_AttitudeSystemCodeGen_old' aborted due to an error.
The selected toolchain ('Pixhawk Toolchain') is not registered. It must be in the Toolchain registry in order to be valid. Check if this toolchain is supported.

RflySim, 2020-10-22:
Hello, we need more information to find out your problem. Please download the latest e0.zip, and compile it to paste all output text in the diagnostic window.

LILILIHANG, 2020-10-22:
> 您好，我们需要更多信息来查找您的问题。请下载最新的e0.zip，然后将其编译以将所有输出文本粘贴到诊断窗口中。

> Hello, we need more information to find out your problem. Please download the latest e0.zip, and compile it to paste all output text in the diagnostic window.

### Starting build procedure for model: Exp4_AttitudeSystemCodeGen
### Build procedure for model: 'Exp4_AttitudeSystemCodeGen' aborted due to an error.
The selected toolchain ('Pixhawk Toolchain') is not registered. It must be in the Toolchain registry in order to be valid. Check if this toolchain is supported. 
For a list of supported factory toolchains refer to https://www.mathworks.com/support/compilers.html