RflySim/RflyExpCode #38: HIL not working with 3DDisplay

amscosta, 2020-10-30:
Hi All,
Following the book guideline, I am uploading the firmware to the pixhawk board for the HIL simulation. 
But the light keeps green blinking. 
I am attaching an image of my screen. I am using a FUTABA T8J radio with the three position switch C configured according to the book. 
Any comment would be very welcome.
Thanks
Alexandre - Brazil
[image]

kcx064, 2020-10-31:
Hi, after conneting your pixhawk with CopterSim,  you can also open QGC to check info from px4 software to help find problem. Checking if have set flight mode binding with RC with QGC, or set px4 to HIL mode.-------- 原始邮件 --------发件人： amscosta <notifications@github.com>日期： 2020年10月31日周六 凌晨2:55收件人： RflySim/RflyExpCode <RflyExpCode@noreply.github.com>抄送： Subscribed <subscribed@noreply.github.com>主    题： [RflySim/RflyExpCode] HIL not working with 3DDisplay (#38)
Hi All,
Following the book guideline, I am uploading the firmware to the pixhawk board for the HIL simulation.
But the light keeps green blinking.
I am attaching an image of my screen. I am using a FUTABA T8J radio with the three position switch C configured according to the book.
Any comment would be very welcome.
Thanks
Alexandre - Brazil

kcx064, 2020-10-31:
Hi,  which firmware did you upload? If your firmware is generated from simulink,  you may switch CH5 on RC to arm px4/pixhawk.

kcx064, 2020-10-31:
You can also use digital data transmission to connect pixhawk with QGC to check info showed on QGC.

amscosta, 2020-11-06:
Thanks a lot for the clue. It worked.

closed 2020-11-06 by amscosta