RflySim/RflyExpCode #41: e0/2.PSPOfficialExps/px4demo_input_rc编译成功，下载情况如下，不知是否属于下载成功，下载后灯不闪，重新上电也不闪，动遥控器没反应

djpv5, 2020-11-02:
Loaded firmware for 9,0, size: 1481437 bytes, waiting for the bootloader...
If the board does not respond within 1-2 seconds, unplug and re-plug the USB connector.
Found board 9,0 bootloader rev 5 on COM6
ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff
Erase  : [====================] 100.0%
Program: [====================] 100.0%
Verify : [                    ] 1.0%
ERROR: timeout waiting for data (4 bytes)

kcx064, 2020-11-02:
确认下下载的时候QGC和CopterSim是关闭的

djpv5, 2020-11-02:
> 确认下下载的时候QGC和CopterSim是关闭的

需要打开这两个软件吗？打开还是一样啊，verify失败啊啊啊

05xichen16, 2022-01-24:
@djpv5 
我也遇到了这个问题，请问是下载了最新的固件吗

SheldonSEU, 2023-05-16:
> @djpv5 我也遇到了这个问题，请问是下载了最新的固件吗

请问解决了吗？解决办法是什么