RflySim/RflyExpCode #45: 半实物仿真，姿态响应误差大的问题

superorange1, 2020-11-14:
请教大家一个问题，我用北航的coptersim半实物仿真出来的响应很差（有很大的误差），相同的参数做gazebo仿真，响应却很好，什么原因？
[image]
[image]
下面是gazebo的，上面是北航CopterSim的。

kcx064, 2020-11-14:
两个的飞机模型是一样的吗

RflySim, 2020-11-14:
你好。我们的模型是充分考虑风阻等因素的高逼真模型，正常情况下做姿态响应时，瞬间姿态可以调整过去，但随着速度的上升，机身会受到空气阻力，等到最终速度稳定时，飞机姿态才能达到受力平衡。Gazebo属于理想模型，没有空气阻力作用。我们的模型和实际实验结果更相近一点，你可以自行观察实飞数据。

superorange1, 2020-11-16:
> 你好。我们的模型是充分考虑风阻等因素的高逼真模型，正常情况下做姿态响应时，瞬间姿态可以调整过去，但随着速度的上升，机身会受到空气阻力，等到最终速度稳定时，飞机姿态才能达到受力平衡。Gazebo属于理想模型，没有空气阻力作用。我们的模型和实际实验结果更相近一点，你可以自行观察实飞数据。

首先非常感谢北航师兄的解答，但是我还是有个问题，百思不得其解，想请教师兄。

我去实际飞行后，看飞行日志确实有一些稳态误差，以下是实际飞行的响应
[image]
另外e0-1SoftwareSimExps的仿真程序和半实物仿真比较接近，响应如下图，但是模型在simulink里我打不开，是不是设计了模块
[image]
而e5.1的仿真程序完全能看懂，但是响应很好，是没有稳态误差的
[image]
其实，我还是不明白，产生这个响应“坑”的主要是啥原因。e5.1的模型是带了空气阻力模型的，如下图。
[image]
按理说，pid的I可以补偿稳态误差，仿真的时候增大I确实可以使“坑”变小，但是不能消除。有什么办法可以补偿这个稳态误差吗？