RflySim/RflyExpCode #47: e8_1_HIL firmware not uploading to the PIXHAWK board

amscosta, 2020-11-20:
Hi All,
I am trying to upload the e_8_1_HIL example but the "PX4 PSP: Upload code to Px4FMU" script is only searching in the "COM3" port.
I was able to upload the e_7_3 example because the scripts searchs in a list of "COMX" ports. 
I am attaching pictures to illustrate my issue.
Any guidance how to solve this issue is very welcome.
Alexandre Costa - Brazil. 
[image]
[image]
[image]

WANGshouming4937, 2020-11-26:
You can configure model like this
[image]

amscosta, 2020-11-27:
Hi,
Thank you very much for the advice.