RflySim/RflyExpCode #48: chapter 8 - HIL -CH5 pwm levels

amscosta, 2020-11-27:
Hi All,
I am trying to simulate the HIL experiments from chapter 8 of prof. Quan book.
In the page 365, it is said :
"Simulation performance
Arm the quadcopter for manual control. Readers could pilot the quadcopter
to a certain altitude, as shown in Fig. 12.12. Then, push three-position switch
CH5 back (closest position from the user) to realize the RTL or push the
switch CH5 forward (farthest position from the user) to make the quadcopter
land."
However I am not having the expected behaviour as described above when I fly with the 3Ddisplay app.
My RC CH5 switch is configured with 1100 ms (farthest position from the user); 1500 ms (middle) ; 1900 ms(closest position from the user).
Any clue to solve the issue is very welcome.
Alex Costa - Brazil

WANGshouming4937, 2020-11-28:
The description in the book is not clear. You can refer to the description below. 
Arm the quadcopter for manual control, and fly the quadcopter to a certain altitude. Then push CH5 from the farthest position to the middle position to realize the RTL. Futher, push CH5 to the closest position, and the quadcopter is still in the return state. This is because there is no switch between return and landing in the state transition conditions. Then push CH5 to the farthest position, the quadrotor re-enters the manual control mode, quickly push the CH5 to the closest position, and the quadrotor enters the landing mode.

amscosta, 2020-12-04:
Perfect. Now it is working. Thank you very much.