RflySim/RflyExpCode #49: Pixhawk faulty condition after unsuccessfull PX4 PSP:Upload code to Px4FMU

amscosta, 2020-12-04:
Hi All,
After a faulty upload of the code from the matlab menu (picture below) the Pixhawk
is in a faulty mode (the bigger led is dead like in the attached zip video).
[image]

[faulty-pixhawk.zip](https://github.com/RflySim/RflyExpCode/files/5644336/faulty-pixhawk.zip)

Any clue how to recover the board state is very welcome.

RflySim, 2020-12-12:
When did you download the toolbox? It seems like a bug half a year before. Have you try to download and use the latest installation package on our website? https://rflysim.com/download

amscosta, 2020-12-17:
Hi,
I just pulled the code available from the github and compiled e8.2HIL. This time I was able to
upload the code to the board. However when I use CopterSim, i cannot arm the board pulling the Yaw stick to the right bottom. The channel 5 switch is being detected properly. Please see the image. Any guidance is very welcome.
[image]
P.S. One more question : When I upload the compiled code from the Matlab menu is the board memory completely erased ? Or there are some kind of "BIOS" software in the board that is kept untouched by the upload. Sorry for this newbie question.

amscosta, 2020-12-18:
The upload of the 8.2HIL code was successfull in one PX4FMU 2.4.6 board. However when I try to do
the same in the other PX4FMU 2.4.6 board, the following message is displayed :
[image]

Any clue how to solve the issue is very welcome.

amscosta, 2020-12-18:
Unfortunatelly, the board is uploading nothing at all. 
[image]

amscosta, 2020-12-18:
The status of the board is like in the following video.

https://user-images.githubusercontent.com/31936244/102663556-db303400-415f-11eb-8fd7-a033435d2290.mp4

RflySim, 2021-03-05:
> Hi,
> I just pulled the code available from the github and compiled e8.2HIL. This time I was able to
> upload the code to the board. However when I use CopterSim, i cannot arm the board pulling the Yaw [...]

The firmware will be erased, but the parameters configured by QGC will be remained.

RflySim, 2021-03-05:
> The upload of the 8.2HIL code was successfull in one PX4FMU 2.4.6 board. However when I try to do
> the same in the other PX4FMU 2.4.6 board, the following message is displayed :
> [image]
> 
> Any clue how to solve the issue is very welcome.

It seems like the hardware problem, you can try again or maybe update the bootloader.