RflySim/RflyExpCode #57: Mission Generation using QGround Control

EL-Wadi, 2021-03-31:
Can we build the controller using your Package and and use the Qground Control to generate the trajectory? How?

RflySim, 2021-03-31:
First, you should choose "no" in the "(10) Whether to block the actuator outputs" option of installation UI, which will enable all PX4 functions.
Then, you can delete the modules (EKF, attitude controller, position controller, etc) you want to replace with the generated controller in the cmake file, or comment out the uORB subscribe and publish codes in the cpp file of these modules.
In the end, you can use Simulink to subscribe and publish signals to replace these modules.

For your case, you can replace part of the function of the PX4 navigator module (get trajectory and send target position point to mc_pos_control module). 
Basic knowledge of the PX4 code framework is required for the above modification. You can visit http://dev.px4.io

EL-Wadi, 2021-04-01:
In real flight what modules are disabled in your package? 
Can we use mavros with your package to do hill with Gazebo?
Another question, if I generate a time based trajectory that is function of time, is the time is initiated (i.e. t=0) when the motors are armed or what?

closed 2021-04-01 by EL-Wadi