RflySim/RflyExpCode #61: Windows Development Environment issues

mengchaoheng, 2021-04-19:
I want to use Rflysim to do drone simulation. I have installed the latest version of the px4 windows development environment PX4.Windows.Cygwin.Toolchain.0.9. Now I want to install the px4 support package of UAVtool, but the support package requires the old version of the tool Chain v0.8, how do I uninstall the new tool chain?

我想使用Rflysim来做无人机仿真，我已经安装了最新版的px4 windows开发环境PX4.Windows.Cygwin.Toolchain.0.9，现在想安装UAVtool的px4支持包，但该支持包需要旧版本的工具链v0.8，请问如何卸载新工具链？

mengchaoheng, 2021-04-19:
[image]

RflySim, 2021-04-19:
我们的安装包会自动安装所有环境，不需要再去安装其他东西。之前安装的UAVtools和Mathworks的代码生成工具箱需要卸载。

mengchaoheng, 2021-04-19:
注意2：如果你针对不同的Pixhawk硬件板子想更换不同的编译命令（例如换成px4-fmu-v5_default），只需要在MATLAB中输入命令：PX4CMD('px4-fmu-v5_default') 或者使用命令：PX4CMD px4-fmu-v5_default

这里的编译命令是不是错了，应该是px4_fmu-v5_default

RflySim, 2021-04-20:
> 注意2：如果你针对不同的Pixhawk硬件板子想更换不同的编译命令（例如换成px4-fmu-v5_default），只需要在MATLAB中输入命令：PX4CMD('px4-fmu-v5_default') 或者使用命令：PX4CMD px4-fmu-v5_default
> 
> 这里的编译命令是不是错了，应该是px4_fmu-v5_default

这里确实有问题，后续我们会修正。

mengchaoheng, 2021-04-20:
戴老师您好，我在发邮件咨询rflysim高级版购买事宜，但官方邮件一直没有回复，可否加相关工作人员联系方式进一步咨询？我电话：17665137013（微信同号）

| |
蒙超恒
|
|
***@***.***
|

签名由 网易邮箱大师 定制

On 04/20/2021 15:55, Xunhua Dai wrote:

注意2：如果你针对不同的Pixhawk硬件板子想更换不同的编译命令（例如换成px4-fmu-v5_default），只需要在MATLAB中输入命令：PX4CMD('px4-fmu-v5_default') 或者使用命令：PX4CMD px4-fmu-v5_default

这里的编译命令是不是错了，应该是px4_fmu-v5_default

这里确实有问题，后续我们会修正。