RflySim/RflyExpCode #62: flightgear卡顿然后直接卡死

mengchaoheng, 2021-04-22:
我在图形工作站上运行基础版，按照https://rflysim.com/zh/3_Using/ControllerDesignAndSimulationPlatform.htm 教程操作，飞机起飞后就flightgear 卡死，请问有什么办法加速运行吗？

[image]

mengchaoheng, 2021-04-22:
一开始安装的时候是运行成功过并且任意操作飞机操作了很长时间的。但后来就不行，重装仿真，重启电脑也不行。不知道什么原因。
[image]

RflySim, 2021-05-30:
FlightGear卡死请确认Simulink是否开启了实时仿真，Simulink的时间应该与实际时钟一致。有可能是你的Simulink仿真速度过快，数据发送速度过于频繁导致卡死。此外，FlightGear卡死的情况下，也可以使用3DDisplay来接收三维数据。

mengchaoheng, 2021-06-17:
确定Simulink是实时，仍然卡顿，都是飞行一会之后才卡。试了3DDisplay，一切ok。请问实际使用中flightgear效果并不好吗？我后期还想使用flightgear自定义机型。网上对flightgear的评价还挺高啊，不知道为何如此卡。

minhoooo1, 2022-01-12:
[image]
在进行这个实验时，飞机一直如下图所示，卡住不动。
[image]

minhoooo1, 2022-01-12:
哦对 是这个报错
[image]

lsw157627, 2022-09-02:
我的也是这个 咋回事  第一次打开

lsw157627, 2022-09-02:
> 哦对 是这个报错 [image]

我的也是这个  咋回事 第一次打开

kukulinxin, 2022-09-29:
> > 哦对 是这个报错[image]
> 
> 我的也是这个 咋回事 第一次打开

解决了吗好兄弟们

mengchaoheng, 2022-09-29:
> 
> 
> 
> 
> 
> 
> > > 哦对 是这个报错[image]
> 
> > 
> 
> > 我的也是这个 咋回事 第一次打开
> 
> 
> 
> 解决了吗好兄弟们

放弃治疗了，我已经不用他们的东西了，太垃圾

kukulinxin, 2022-09-29:
> 

有无其他仿真软件推荐啊好兄弟，身边还没实物，谢谢

kukulinxin, 2022-09-29:
> > > > 哦对 是这个报错[image]
> > 
> > 
> > > 
> > 
> > 
> > > 我的也是这个 咋回事 第一次打开
> > 
> > 
> > 解决了吗好兄弟们
> 
> 放弃治疗了，我已经不用他们的东西了，太垃圾

有无其他仿真软件推荐啊好兄弟，身边还没实物，谢谢

JohnRen325, 2023-03-11:
尝试一下关闭Windows的防火墙和杀毒软件，我关闭了域网络、专业网络和公用网络的防火墙，再运行仿真，就可以正常飞行了。