RflySim/RflyExpCode #76: About the test case program

SasakiManabu, 2021-11-12:
I don’t know the contact information, so I wrote it here.

I have some questions about this video "https://www.youtube.com/watch?v=-M9OMHFWO4w".

1.Do you make test case programs with Simulink?
2.When setting up test case in a test case program, are you sending µORB message to CopterSim via UDP?

SasakiManabu, 2021-11-18:
If the contact information for the question is different, please let me know where to contact you.

RflySim, 2021-11-18:
We have released the RflySim advanced version for free, which has the fault injection function and cases. However, the document only has a Chinese version now. You can download it from here https://rflysim.com/download.html and get the demo at RflySimAPIs\FaultInjectionAPI. 
Currently, we send our self-defined UDP struct to CopterSIm via UDP to trigger the model faults like wind, sensor errors. Of course, we also have API to send uORB msg to PX4, which you can use to trigger the controller faults.

closed 2021-11-18 by SasakiManabu