RflySim/RflyExpCode #88: PSP工具箱配置错误

WyvernLeague, 2022-03-18:
[image]
Warning: PX4PSP toolbox failed to refresh MATLAB toolbox...
错误: PSP工具箱配置失败，请重试！
错误: PSP工具箱升级更新出错，安装中止

RflySim, 2022-03-18:
你好。请重启电脑后再尝试。可能是因为电脑改动了“我的文档”所在盘符。

WyvernLeague, 2022-03-19:
反复尝试，还是不行。我确认一下哈，打开ISO文件之后，会产生DVD驱动器G，选择此作为路径应该没错吧。
[image]

zhengzhiZJU, 2022-08-16:
[image]
[image]
[image]
一样的问题，我是win11