RflySim/RflyExpCode #138: 启动SITL时出现没有wsl子系统报错

yyyyyxq, 2024-05-13:
[image]
[image]
[image]

使用exe安装后，启动SITL时出现没有wsl子系统报错。但是实际上已经安装也并未禁用，版本也降为1.

RflySim, 2024-05-13:
不是平台的问题，而是你的CMD环境变量坏掉了。导致bat脚本命令执行错误。

网上搜一下，修复 系统环境变量 的方法。

yyyyyxq, 2024-05-13:
> 不是平台的问题，而是你的CMD环境变量坏掉了。导致bat脚本命令执行错误。
> 
> 网上搜一下，修复 系统环境变量 的方法。

已经恢复初始设置，仍然不行，我将重新编译安装。另外编译时出现 错误：无效类，会对后续有影响么？
[image]